FTC-9073-Knightrix/2018-19-Knightrix #12: CAD the intake

colettedcosta, 2018-10-20:


ibravo, 2018-10-20:
@Tkspins Once you finish modeling, please add the files to this ticket for me to print. Thanks!

ibravo, 2018-10-25:
Intake was printed on Monday and side parts are printed as of today. Please use today's meeting to test intake and finalize documentation.

Tkspins, 2018-10-27:
@ibravo I attached the zip file with the assembly and all .stl and .ipt files. Only the sides and sides_MIR needs to be printed. 
[Intake Print 10.27.18.zip](https://github.com/FTC-9073-Knightrix/2018-19-Knightrix/files/2521892/Intake.Print.10.27.18.zip)

ibravo, 2018-10-29:
[2018.10.27.Intake.Carousel.zip](https://github.com/FTC-9073-Knightrix/2018-19-Knightrix/files/2527373/2018.10.27.Intake.Carousel.zip)

Tony, intake Inventor files are attached. Make the mods you need.

Tkspins, 2018-11-02:
Here is the updated Inventor file. Please let me know if any other updates are needed. 
[Updated Intake Carousel.zip](https://github.com/FTC-9073-Knightrix/2018-19-Knightrix/files/2540925/Updated.Intake.Carousel.zip)

ibravo, 2018-11-02:
@Tkspins  Sorry, didn't realize that the file did not include de Inventor format. Here they are. 
I saw your design, but we need an intake where the walls are not parallel, but rather wider at the end to capture more minerals. Like a V shape.
The top section, where the axle for the motor is located, should be parallel though.
 
[2018.10.27.Intake.Carousel.zip](https://github.com/FTC-9073-Knightrix/2018-19-Knightrix/files/2542163/2018.10.27.Intake.Carousel.zip)

Tkspins, 2018-11-03:
@ibravo I attached the updated files. Sorry it took so long, the update for inventor was very slow. 

[Updated Intake 2.zip](https://github.com/FTC-9073-Knightrix/2018-19-Knightrix/files/2544544/Updated.Intake.2.zip)

ibravo, 2018-11-03:
Thanks @Tkspins . I'll put it through the 3D printer now.

Tkspins, 2018-11-04:
Here is the intake with separate pieces. It won't let me upload to Autodesk Desktop Connector for some reason. 
[Intake - Separate Pieces.zip](https://github.com/FTC-9073-Knightrix/2018-19-Knightrix/files/2545566/Intake.-.Separate.Pieces.zip)

ibravo, 2018-12-04:
Closing this issue as a result of scraping the carousel intake and going with a simpler rubber band design.

closed 2018-12-04 by ibravo