FTC-9073-Knightrix/2018-19-Knightrix #9: ARNAV NEEDS TO OPEN UP THE DRIVETRAIN BOX AND BUILD IT

vjay49, 2018-10-20:


teenymermaid, 2018-10-20:
who is vijay

teenymermaid, 2018-10-20:
vijay u finish the intake

teenymermaid, 2018-10-20:
vijay u cant build

teenymermaid, 2018-10-20:
http://files.andymark.com/AssemblyInstructions/am-3437_HDmec+TileRunner+HDMecanum+User+Guide.pdf

skinariwala, 2018-10-20:
Need pieces. Location in the engineering meeting entry.

ibravo, 2018-10-20:
@vraj123 Can you please add the pieces that we need to buy to this ticket so Mr Baer can complete the purchase? 
Thanks

vraj123, 2018-10-20:
Parts needed:  16 Pico  Spacer  (am-3406)(http://www.andymark.com/product-p/am-3406.htm)  and 8 63mm Churro (am-3399) (https://www.andymark.com/product-p/am-churro.htm)

ibravo, 2018-12-10:
Instructions on the Drive Train purchased:
http://files.andymark.com/AssemblyInstructions/am-3437_HDmec+TileRunner+HDMecanum+User+Guide.pdf

closed 2018-12-10 by ibravo